leaphy-robotics/leaphy-webbased-svelte #45: [bug] some svgs don't load?

koen1711, 2024-05-05:
### What happened?

[image]

### Version

latest

### What browsers are you seeing the problem on?

Firefox, Chrome, Safari, Microsoft Edge, Other

### Relevant log output

```shell
Failed to load resource: the server responded with a status of 404 (Not Found)
l_original_nano_rp2040_actuators.svg:1
```

koen1711, 2024-05-05:
Same here.
[image]

rmoesbergen, 2024-05-05:
This is a bug in the blockly repo, the images are not present there either