leaphy-robotics/leaphy-webbased-svelte #160: [bug]

beekumo, 2025-01-18:
### What happened?

LED block for Leaphy original is now on pin 8, 9, 10 and 11 should be pin 8, 7, 6, 5 because pin 11 is in use for DC motor. Please note that Led block pin layout needs to remain the same for Leaphy Flitz and Starling.

### Version

latest

### Relevant log output

```shell

```

rmoesbergen, 2025-01-19:
analogWrite does this on a non-pwm pin:
```
			case NOT_ON_TIMER:
			default:
				if (val < 128) {
					digitalWrite(pin, LOW);
				} else {
					digitalWrite(pin, HIGH);
				}
```
So that should "just work" if we only change the pins. Ofcourse it doesn't allow for gradual colors anymore on that pin...

rmoesbergen, 2025-01-19:
Hmm:
```
        if (arduino.robotType.includes("nano")) {
            pin_red = 11;
            pin_green = 10;
            pin_blue = 9;
            arduino.addSetup(
                "setup_nano_rgb",
                "pinMode(8, OUTPUT);\n  digitalWrite(8, LOW);",
                false,
            );
        } else {
            pin_red = 3;
            pin_green = 5;
            pin_blue = 6;
        }
```

So for Uno the pins are 3,5,6, not 8,9,10,11... ?

rmoesbergen, 2025-01-19:
This cannot be fixed until we have a 'Starling' robot type. Blocked by #161

rmoesbergen, 2025-03-27:
Pin 8, 7, 6 en 5 zouden rgb moeten zijn met 8 als ground:

[image]